MIT-SPARK/Kimera-VIO #157: Customize dataprovider.cpp/Eurocdataprovider.cpp for own dataset

bghadge, 2021-04-21:
I have my own dataset and I want to try Kimera-VIO on it. It seems like it will be a bit difficult to create my own dataprovider, so I converted my dataset into the euroc dataset format. Currently, I'm trying to understand how I can "turn off" the parseGTdata function in the eurocdataprovider, since I do not have the ground truth data. Could you please let me know how to modify eurocdataprovider.cpp for custom datasets?

marcusabate, 2022-09-28:
I would recommend searching in that file for any references to parseGTdata (most likely in the spin) and commenting them out. It should just be done by commenting out the call to that function or whichever function is responsible for actually reading from the file.

There may be parameters setup that point to GT data files, which you'll have to remove. You can also look for calls to CHECK that are associated with GT data, I expect there are a few intended to make sure that the data has the correct size. If you iterate between commenting out GT code and building, checking for the error and then commenting out the relevant code, you'll get there pretty quick.

closed 2022-09-28 by marcusabate

HtutLynn, 2023-10-25:
Hi, I am sorry for re-opening the issue. I was wondering if there are any update to this? I am trying to convert sensor data from iOS to EuroC format and trying to use Kimeara-VIO for generating global trajectories.

HtutLynn, 2023-10-25:
I tried removing any references to `parseGTdata` function and also all `CHECK` calls to GT data from EurocDataProvider.cpp but new error from GTSAM is causing me unable to run Kimeara-VIO. I am testing with just EUROC dataset and it's not working anymore after removing all GT related calls.

```
I1025 10:37:58.958392  4880 EurocDataProvider.cpp:804] Running dataset between frame 50 and frame 2912
terminate called after throwing an instance of 'gtsam::IndeterminantLinearSystemException'
  what():
Indeterminant linear system detected while working near variable
8646911284551352320 (Symbol: x0).

Thrown when a linear system is ill-posed.  The most common cause for this
error is having underconstrained variables.  Mathematically, the system is
underdetermined.  See the GTSAM Doxygen documentation at
http://borg.cc.gatech.edu/ on gtsam::IndeterminantLinearSystemException for
more information.
```

marcusabate, 2023-11-03:
@HtutLynn Discussion moved to #218.