MIT-SPARK/Kimera-VIO #160: Kimera VIO Mono on custom data

Alex-Dee, 2021-04-27:
Hi,
I have successfully build and run Kimera-VIO Mono Pipeline on the EuRoC dataset and the results look very promising. However when I switch to either a custom or other scientific data set such as UMA-VI the algorithm diverges and the estimated poses reach escape velocity quickly (see attached image). Does anybody else experience these kind of problems? It seems to me that VIO is super parameter sensitive and not robust at all. Has anybody successfully processed custom data and may be able to share some hints on how to proceed? That would be tremendously helpful.
[image]

mateosss, 2021-09-02:
I opened a similar issue #170. Did you achieve any improvements?

Alex-Dee, 2021-09-08:
> I opened a similar issue #170. Were you able to achieve any improvements?

Sadly I was not able to identify the problem even thorugh extensive debugging. Running the same data through ORB-SLAM3 went nicely and without any problems. So I am pretty sure it's a problem either with the code or my specific build environment.

zgxsin, 2022-07-07:
I had the same issue. I checked the performance of Kimera on Euro dataset MH05 using both stereo and mono setup without any modifications. I got mean APE error of 0.17 and 0.95 respectively. But when i used Kimera on my customer datasets, the result didn't make any sense.

marcusabate, 2022-09-28:
Note that our paper results with Euroc were always in the stereo configuration, mono is relatively new and untested. However, we have found that applying the change I mentioned in #170 improves performance significantly especially if your custom dataset is not time-synchronized between imu and camera.

For Euroc, we've found that mono performs worse than stereo but not by too much. There is a separate parameters folder for the Mono case, generally has to be looser to work properly.

closed 2023-11-03 by marcusabate