MIT-SPARK/Kimera-VIO #169: Comparisson with ORB-SLAM2 and 3

mateosss, 2021-07-06:
Hello, is there any up-to-date comparison between the current Kimera VIO capabilities and ORB-SLAM2 or ORB-SLAM3?
The only one I could find was in the [ORB-SLAM3 paper](https://arxiv.org/pdf/2007.11898.pdf#page=12) which showed favorable results for ORB-SLAM3. Are these metrics still relevant?

emanuelazfernandes, 2022-05-06:
@mateosss afaik, ORB-SLAM3 is just for the VIO-SLAM, while Kimera is focused on the 3D Scene graphs hierarchical representation. The author's group has more recent work towards real-time optimizations, but so far I don't think there is code available: https://arxiv.org/abs/2201.13360
I highly recommend this author's presentation for a good overview:
https://www.youtube.com/watch?v=TUOCMevmbOg

mateosss, 2022-05-06:
Thanks for the links @emanuelazfernandes! Very interesting talk so far.

For anyone else seeing this issue that's mostly interested in SLAM/VIO tracking, the metrics reported in the ORB-SLAM3 paper are still about right. Since then I tried Kimera-VIO but it seemed to not be very robust and datasets outside the standard EuRoC ones usually resulted in divergence and crashes. ORB-SLAM3 was better but its trajectory is very jittery which is a big downside for my target application (XR). Finally I settled on Basalt which, while only being able to perform VIO on realtime and thus achieving worse ATE, it has the shortest runtimes, smoothest trajectories (better RTE), seems to be the most stable (except when your device has frame drops), and has a permissive non-GPL license.

closed 2022-05-06 by mateosss