MIT-SPARK/Kimera-VIO #170: Kimera MonoImuPipeline drift on Realsense D455

mateosss, 2021-08-09:
Hello, I'm trying to use Kimera mono-inertial with a Realsense D455 using the left infrared camera 640x360 15fps. While the rotations seem to be properly reflected the trajectory ends up drifting a lot.

[Here](https://drive.google.com/file/d/1F08zNvRgFBDS4CIh1GhC2DaD7pr9aPxC/view?usp=sharing) is an example of the problem as a EuRoC-formatted dataset, it was run with these [parameter files](https://github.com/MIT-SPARK/Kimera-VIO/files/6951743/D455_params.zip).
It should be enough to modify `DATASET_PATH` and `PARAMS_PATH` in `stereoVIOEuroc.bash` to run the dataset.
In any case [here](https://youtu.be/RDS7kItlo_s) is a video of the problem (though with some delay between frames to better see the problem).

Do you have any idea what the cause of the drift could be? This happens without any modifications to Kimera.

Versions used:
- Kimera-VIO: current master MIT-SPARK/Kimera-VIO@641576fd86bdecbd663b4db3cb068f49502f3a2c
- GTSAM: borglab/gtsam@37ea955d2cf8e1651a01934d1363966b7e4404c9
- OpenGV: current master laurentkneip/opengv@91f4b19c73450833a40e463ad3648aae80b3a7f3 
- OpenCV: 3.3.1
- OS: Ubuntu 20.04

mateosss, 2021-08-10:
Just an update that might be useful. I noticed my D455 accelerometer gravity is a bit different on each axis.
```python
+X ~= 9.81, -X ~= 9.80
+Y ~= 9.79, -Y ~= 9.84
+Z ~= 9.97, -Z ~= 9.62
```

Could that difference between the axes' gravity values (mainly the Z-axis) be causing this drift?

duda1202, 2022-06-28:
Were you able to tune it properly? I am having the same issue trying to use the MonoImu pipeline. Even with EurocMono on the euroc dataset, it drifts rapidly.

zgxsin, 2022-07-06:
> No, not really. After a while I moved on to other systems, I figured Kimera's strengths are not on the tracking aspect.
> 
> ORB-SLAM3 has pretty nice features including monocular tracking, but it's very unstable.
> 
> Basalt proved to be pretty good for stereo VIO and also has a permissive license. Although it doesn't support monocular tracking out of the box (there is [this](https://github.com/DLR-RM/granite) fork of Basalt that is monocular though).
> 
> For monocular tracking, DM-VIO looks amazing on paper, but I haven't had the time to try it out. And it's GPL.

Did you try openvins/vins_mono? It has nice results. But the license is GPL

marcusabate, 2022-09-28:
If you are using IMU this could be caused by the imu interpolation. We're pushing an update soon that should fix this. If you want the fix quickly, change [this line](https://github.com/MIT-SPARK/Kimera-VIO/blob/641576fd86bdecbd663b4db3cb068f49502f3a2c/src/dataprovider/DataProviderModule.cpp#L58) from `getImuDataInterpolatedUpperBorder` to `getImuDataInterpolatedBorders`. With this change I can run mono very effectively using Kimera.

If it doesn't work I'm happy to help further. I can say that Vins-Fusion does perfom well in the mono configuration as well.

mateosss, 2022-10-01:
Thanks for the update @marcusabate I'll be sure to try it out at some point and update this issue. If I can't make it should be verifiable with just the files provided in the description.

wolfcanli, 2022-11-12:
> Thanks for the update @marcusabate I'll be sure to try it out at some point and update this issue. If I can't make it should be verifiable with just the files provided in the description.

Hi, any good news for mono-imu mode? Thank you.

marcusabate, 2023-10-04:
This should be resolved in the `feature/prerelease` branch, where we've included the necessary update. Let me know if this is still an issue. Closing for now because this is an old issue but feel free to reopen.

closed 2023-10-04 by marcusabate