MIT-SPARK/Kimera-VIO #183: GncOptimizer.h: No such file or directory

Z-Jeff, 2022-01-14:
When running "catkin build", it occurs an error: 
**catkin_ws/src/Kimera-RPGO/src/RobustSolver.cpp:16:10: fatal error: gtsam/nonlinear/GncOptimizer.h: No such file or directory**
It seems to be "gtsam" version problem, but I have no ideas to solve this.
How can I fix this?
My "gtsam" version is 4.1.0, using Ubuntu20.04 and Ros-Noetic.

kanster, 2022-03-07:
Hi Jeff,

You need to checkout to a specific version of gtsam, if I remember it correctly, the name of the branch is related with **gncimprovements**

BR

closed 2022-09-28 by marcusabate

zhuoyuan00, 2023-11-30:
Hi Kanster,
The gtsam branch relevant to gncimprovements can not been find. Could you please show more infomation for the right gtsam version?
By the way mine is 4.0.3, using Ubuntu18. This problem also occurs when I was installing RPGO.
Could you help me?
Thanks very much.