MIT-SPARK/Kimera-VIO #186: Docker build fails

sanderfo, 2022-03-02:
**Description:**
Hi. Tried to build on a fresh Ubuntu 18.04 machine, with a fresh docker install.  The docker build fails, and I think it's due to a recent commit in the GTSAM master branch. The suspected commit removed some members that Kimera seems to use. The suspected commit is here: https://github.com/borglab/gtsam/commit/bcbd26c4fbc443dbef8823ca62df17853d3e5ad6

**Command:**
```
sudo docker build --rm -t kimera_vio -f ./scripts/docker/Dockerfile . 

```

**Console output:**
```
Copied the last part:

[ 31%] Building CXX object CMakeFiles/kimera_vio.dir/src/factors/PointPlaneFactor.cpp.o
/root/Kimera-VIO/src/backend/VioBackendParams.cpp: In static member function 'static void VIO::BackendParams::setIsam2Params(const VIO::BackendParams&, gtsam::ISAM2Params*)':
/root/Kimera-VIO/src/backend/VioBackendParams.cpp:46:15: error: 'struct gtsam::ISAM2Params' has no member named 'setCacheLinearizedFactors'; did you mean 'cacheLinearizedFactors'?
   isam_param->setCacheLinearizedFactors(true);
               ^~~~~~~~~~~~~~~~~~~~~~~~~
               cacheLinearizedFactors
/root/Kimera-VIO/src/backend/VioBackendParams.cpp:51:15: error: 'struct gtsam::ISAM2Params' has no member named 'setEvaluateNonlinearError'; did you mean 'evaluateNonlinearError'?
   isam_param->setEvaluateNonlinearError(false);  // only for debugging
               ^~~~~~~~~~~~~~~~~~~~~~~~~
               evaluateNonlinearError
In file included from /root/Kimera-VIO/include/kimera-vio/backend/RegularVioBackend.h:28:0,
                 from /root/Kimera-VIO/include/kimera-vio/backend/VioBackendModule.h:18,
                 from /root/Kimera-VIO/src/backend/VioBackendModule.cpp:16:
/root/Kimera-VIO/include/kimera-vio/backend/VioBackend.h: In member function 'void VIO::VioBackend::saveGraph(const string&) const':
/root/Kimera-VIO/include/kimera-vio/backend/VioBackend.h:183:65: warning: 'void gtsam::NonlinearFactorGraph::saveGraph(std::ostream&, const gtsam::Values&, const gtsam::GraphvizFormatting&, const KeyFormatter&) const' is deprecated [-Wdeprecated-declarations]
     smoother_->getFactors().saveGraph(ofstream_wrapper.ofstream_);
                                                                 ^
In file included from /usr/local/include/gtsam_unstable/nonlinear/FixedLagSmoother.h:25:0,
                 from /usr/local/include/gtsam_unstable/nonlinear/IncrementalFixedLagSmoother.h:23,
                 from /root/Kimera-VIO/include/kimera-vio/backend/RegularVioBackend-definitions.h:25,
                 from /root/Kimera-VIO/include/kimera-vio/backend/RegularVioBackend.h:25,
                 from /root/Kimera-VIO/include/kimera-vio/backend/VioBackendModule.h:18,
                 from /root/Kimera-VIO/src/backend/VioBackendModule.cpp:16:
/usr/local/include/gtsam/nonlinear/NonlinearFactorGraph.h:280:27: note: declared here
     void GTSAM_DEPRECATED saveGraph(
                           ^~~~~~~~~
In file included from /root/Kimera-VIO/src/backend/VioBackend.cpp:30:0:
/root/Kimera-VIO/include/kimera-vio/backend/VioBackend.h: In member function 'void VIO::VioBackend::saveGraph(const string&) const':
/root/Kimera-VIO/include/kimera-vio/backend/VioBackend.h:183:65: warning: 'void gtsam::NonlinearFactorGraph::saveGraph(std::ostream&, const gtsam::Values&, const gtsam::GraphvizFormatting&, const KeyFormatter&) const' is deprecated [-Wdeprecated-declarations]
     smoother_->getFactors().saveGraph(ofstream_wrapper.ofstream_);
                                                                 ^
In file included from /usr/local/include/gtsam/nonlinear/ISAM2Result.h:28:0,
                 from /usr/local/include/gtsam/nonlinear/ISAM2.h:26,
                 from /root/Kimera-VIO/include/kimera-vio/backend/VioBackend.h:41,
                 from /root/Kimera-VIO/src/backend/VioBackend.cpp:30:
/usr/local/include/gtsam/nonlinear/NonlinearFactorGraph.h:280:27: note: declared here
     void GTSAM_DEPRECATED saveGraph(
                           ^~~~~~~~~
In file included from /root/Kimera-VIO/src/backend/VioBackend.cpp:30:0:
/root/Kimera-VIO/include/kimera-vio/backend/VioBackend.h: In constructor 'VIO::VioBackend::VioBackend(const gtsam::Pose3&, const StereoCalibPtr&, const VIO::BackendParams&, const VIO::ImuParams&, const VIO::BackendOutputParams&, bool)':
/root/Kimera-VIO/include/kimera-vio/backend/VioBackend.h:433:29: warning: 'VIO::VioBackend::backend_output_params_' will be initialized after [-Wreorder]
   const BackendOutputParams backend_output_params_;
                             ^~~~~~~~~~~~~~~~~~~~~~
/root/Kimera-VIO/include/kimera-vio/backend/VioBackend.h:191:29: warning:   'std::atomic<VIO::VioBackend::BackendState> VIO::VioBackend::backend_state_' [-Wreorder]
   std::atomic<BackendState> backend_state_;
                             ^~~~~~~~~~~~~~
/root/Kimera-VIO/src/backend/VioBackend.cpp:68:1: warning:   when initialized here [-Wreorder]
 VioBackend::VioBackend(const Pose3& B_Pose_leftCam,
 ^~~~~~~~~~
In file included from /root/Kimera-VIO/include/kimera-vio/backend/VioBackend-definitions.h:21:0,
                 from /root/Kimera-VIO/include/kimera-vio/backend/VioBackend.h:50,
                 from /root/Kimera-VIO/src/backend/VioBackend.cpp:30:
/root/Kimera-VIO/src/backend/VioBackend.cpp: In member function 'void VIO::VioBackend::addLandmarksToGraph(const LandmarkIds&)':
/root/Kimera-VIO/src/backend/VioBackend.cpp:402:34: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
       LOG_IF(FATAL, obs_kf.first != curr_kf_id_)
                     ~~~~~~~~~~~~~^~~~
In file included from /root/Kimera-VIO/include/kimera-vio/common/VioNavState.h:27:0,
[... 74 lines cut ...]
                 from /root/Kimera-VIO/include/kimera-vio/backend/VioBackend-definitions.h:28,
                 from /root/Kimera-VIO/include/kimera-vio/backend/VioBackend.h:50,
                 from /root/Kimera-VIO/src/backend/VioBackend.cpp:30:
/root/Kimera-VIO/include/kimera-vio/common/vio_types.h: At global scope:
/root/Kimera-VIO/include/kimera-vio/common/vio_types.h:70:13: warning: 'void VIO::getValidKeypointsFromStatusKeypointsCV(const StatusKeypointsCV&, VIO::KeypointsCV*)' defined but not used [-Wunused-function]
 static void getValidKeypointsFromStatusKeypointsCV(
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMakeFiles/Makefile2:163: recipe for target 'CMakeFiles/kimera_vio.dir/all' failed
make[1]: *** [CMakeFiles/kimera_vio.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
The command '/bin/sh -c cd Kimera-VIO && mkdir build && cd build && cmake .. && make -j$(nproc)' returned a non-zero code: 2

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* Kimera-Vio branch, tag or commit used
* GTSAM version used:
* OpenGV version used:
* OpenCV version used: type `opencv_version`
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* Did you change the source code? (yes / no):

sanderfo, 2022-03-02:
I created a pull request for this in https://github.com/MIT-SPARK/Kimera-VIO/pull/187 where I have addressed this issue.

ShrutheeshIR, 2022-04-21:
Hi @sanderfo  I face the same issue, I looked at your PR. But, it looks like your dockerfile has reverted to using this repository. Could you shed some light on this?

marcusabate, 2022-09-28:
@ShrutheeshIR See my comment on #187 [here](https://github.com/MIT-SPARK/Kimera-VIO/pull/187#issuecomment-1260269209). Soon we will update Kimera-VIO to fix this issue (among others), but until then you can either downgrade gtsam to before the break or apply the patch in #187.

closed 2022-11-29 by marcusabate