MIT-SPARK/Kimera-VIO #189: Getting some errors while building

abhidaberao, 2022-04-17:
On running `cmake ..`, I get the following.

```
-- ==============================================================
-- ====================  Dependencies ===========================
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: filesystem serialization regex timer date_time thread
-- Found installed version of gflags: /usr/lib/x86_64-linux-gnu/cmake/gflags
-- Detected gflags version: 2.2.2
-- Failed to find installed glog CMake configuration, searching for glog build directories exported with CMake.
-- Failed to find an installed/exported CMake configuration for glog, will perform search for installed glog components.
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.65") found components: serialization system filesystem thread program_options date_time timer chrono regex
-- GTSAM include directory:  /usr/lib/cmake/GTSAM/../../../include
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0")
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.65") found components: serialization system filesystem thread program_options date_time timer chrono regex
-- GTSAM include directory:  /usr/lib/cmake/GTSAM/../../../include
-- GTSAM include directory:  /usr/lib/cmake/GTSAM/../../../include
CMake Warning at cmake/VerifyGtsamConfig.cmake:48 (message):
  Detected that the GTSAM version to be used is not configured to use
  on-manifold IMU preintegration.  You should build GTSAM with
  GTSAM_TANGENT_PREINTEGRATION=OFF.  You can do this when using catkin by
  cleaning gtsam and running 'catkin config -a --cmake-args
  -DGTSAM_TANGENT_PREINTEGRATION=OFF' and then rebuilding
Call Stack (most recent call first):
  CMakeLists.txt:51 (verify_gtsam_config)

CMake Warning at cmake/VerifyGtsamConfig.cmake:52 (message):
  You can disable warnings related to GTSAM by setting
  KIMERA_VERIFY_GTSAM_CONFIG=OFF in the future.
Call Stack (most recent call first):
  CMakeLists.txt:51 (verify_gtsam_config)

-- Vocabulary file exists, will not download.
-- Pangolin not found.
-- ===============================================================
-- ================  Configuration Options  ======================
-- CMAKE_CXX_COMPILER_ID type                : GNU
-- CMAKE_CXX_COMPILER_VERSION                : 9.4.0
-- Build flags
--   Build type                              : Release
--   C compilation flags (Release)           :  -O3 -DNDEBUG
--   C++ compilation flags (Release)         :  -O3 -DNDEBUG
-- Configuring done
-- Generating done
-- Build files have been written to: /home/abhishek/Downloads/Kimera-VIO
```

pakshi10, 2022-05-09:
I am getting the same error.

mikekung108, 2022-05-30:
The problem is the version of **GTSAM**. So, 
``` console
# go to gtsam file
git checkout 4.1.1
```
then re-build and re-install again, everything should be fine.

jaku-jaku, 2022-09-22:
Latest gtsam requires Pangolin from Cmakelist response.
You can install Pangolin, but Pangolin has some minor issue, fixed in https://github.com/stevenlovegrove/Pangolin/pull/792

marcusabate, 2022-09-28:
I am able to run 20.04 (native) using gtsam commit `e5866799dff48239573cdd84964180867e50edd2`. Latest gtsam requires modifications to the source code, but an update to Kimera-VIO soon will bring those in. Refer to #194 for more info if you want to use a more recent gtsam.