MIT-SPARK/Kimera-VIO #192: error on line DEFINE_bool(visualize_feature_predictions,

fatmanurozdemir, 2022-07-22:
**error: expected constructor, destructor, or type conversion before ‘(’ token
   30 | DEFINE_bool(visualize_feature_predictions**

I am getting this error while building. I guess it is related to gflags library.
I built the gflags lib from source and also using sudo apt-get install libgflags-dev command. I also added 
list(APPEND CMAKE_PREFIX_PATH <gflags-config.cmake path>)
line into CMakeLists.txt but it did not work.

**Command:**
```
make -j $(nproc)```

```

Consolidate compiler generated dependencies of target gtest
[  2%] Built target gtest
Consolidate compiler generated dependencies of target kimera_vio
[  3%] Building CXX object CMakeFiles/kimera_vio.dir/src/frontend/StereoImuSyncPacket.cpp.o
[  4%] Building CXX object CMakeFiles/kimera_vio.dir/src/frontend/StereoVisionImuFrontend.cpp.o
[  5%] Building CXX object CMakeFiles/kimera_vio.dir/src/frontend/MonoImuSyncPacket.cpp.o
[  5%] Building CXX object CMakeFiles/kimera_vio.dir/src/frontend/MonoVisionImuFrontend.cpp.o
[  6%] Building CXX object CMakeFiles/kimera_vio.dir/src/frontend/VisionImuFrontend.cpp.o
[  7%] Building CXX object CMakeFiles/kimera_vio.dir/src/frontend/VisionImuFrontendModule.cpp.o
[  8%] Building CXX object CMakeFiles/kimera_vio.dir/src/frontend/Tracker.cpp.o
[  9%] Building CXX object CMakeFiles/kimera_vio.dir/src/frontend/feature-detector/FeatureDetectorParams.cpp.o
In file included from /mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/include/kimera-vio/utils/UtilsOpenCV.h:34,
                 from /mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/include/kimera-vio/frontend/CameraParams.h:27,
                 from /mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/include/kimera-vio/frontend/Frame.h:36,
                 from /mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/include/kimera-vio/frontend/StereoFrame.h:40,
                 from /mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/include/kimera-vio/frontend/VisionImuFrontendParams.h:29,
                 from /mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/src/frontend/feature-detector/FeatureDetectorParams.cpp:16:
/mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/include/kimera-vio/common/vio_types.h:70:13: warning: ‘void VIO::getValidKeypointsFromStatusKeypointsCV(const StatusKeypointsCV&, VIO::KeypointsCV*)’ defined but not used [-Wunused-function]
   70 | static void getValidKeypointsFromStatusKeypointsCV(
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 10%] Building CXX object CMakeFiles/kimera_vio.dir/src/frontend/feature-detector/FeatureDetector.cpp.o
/mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/src/frontend/Tracker.cpp:30:12: error: expected constructor, destructor, or type conversion before ‘(’ token
   30 | DEFINE_bool(visualize_feature_predictions,
      |            ^
In file included from /mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/src/frontend/Tracker.cpp:8:
/mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/include/kimera-vio/frontend/Tracker.h: In constructor ‘VIO::Tracker::Tracker(const VIO::FrontendParams&, const ConstPtr&, VIO::DisplayQueue*)’:
/mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/include/kimera-vio/frontend/Tracker.h:155:14: warning: ‘VIO::Tracker::landmark_count_’ will be initialized after [-Wreorder]
  155 |   LandmarkId landmark_count_;
      |              ^~~~~~~~~~~~~~~
/mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/include/kimera-vio/frontend/Tracker.h:61:24: warning:   ‘const VIO::FrontendParams VIO::Tracker::tracker_params_’ [-Wreorder]
   61 |   const FrontendParams tracker_params_;
      |                        ^~~~~~~~~~~~~~~
/mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/src/frontend/Tracker.cpp:37:1: warning:   when initialized here [-Wreorder]
   37 | Tracker::Tracker(const FrontendParams& tracker_params,
      | ^~~~~~~
/mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/src/frontend/Tracker.cpp: In member function ‘void VIO::Tracker::featureTracking(VIO::Frame*, VIO::Frame*, const gtsam::Rot3&, boost::optional<cv::Mat>)’:
/mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/src/frontend/Tracker.cpp:155:31: warning: comparison of integer expressions of different signedness: ‘const size_t’ {aka ‘const long unsigned int’} and ‘const int’ [-Wsign-compare]
  155 |     if (!status[i] || lmk_age > tracker_params_.maxFeatureAge_) {
      |                       ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/src/frontend/Tracker.cpp:180:25: error: ‘FLAGS_visualize_feature_predictions’ was not declared in this scope
  180 |   if (display_queue_ && FLAGS_visualize_feature_predictions) {
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/src/frontend/Tracker.cpp: In member function ‘std::pair<VIO::TrackingStatus, gtsam::Pose3> VIO::Tracker::geometricOutlierRejectionMono(VIO::Frame*, VIO::Frame*)’:
/mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/src/frontend/Tracker.cpp:245:36: warning: comparison of integer expressions of different signedness: ‘std::vector<int>::size_type’ {aka ‘long unsigned int’} and ‘const int’ [-Wsign-compare]
  245 |   if (mono_ransac_.inliers_.size() < tracker_params_.minNrMonoInliers_) {
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/src/frontend/Tracker.cpp: In member function ‘std::pair<VIO::TrackingStatus, gtsam::Pose3> VIO::Tracker::geometricOutlierRejectionMonoGivenRotation(VIO::Frame*, VIO::Frame*, const gtsam::Rot3&)’:
/mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/src/frontend/Tracker.cpp:338:46: warning: comparison of integer expressions of different signedness: ‘std::vector<int>::size_type’ {aka ‘long unsigned int’} and ‘const int’ [-Wsign-compare]
  338 |   if (mono_ransac_given_rot_.inliers_.size() <
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
  339 |       tracker_params_.minNrMonoInliers_) {
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~       
/mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/src/frontend/MonoVisionImuFrontend.cpp: In member function ‘VIO::MonoFrontendOutput::UniquePtr VIO::MonoVisionImuFrontend::nominalSpinMono(VIO::MonoImuSyncPacket::UniquePtr&&)’:
/mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/src/frontend/MonoVisionImuFrontend.cpp:96:8: warning: variable ‘tic_full_preint’ set but not used [-Wunused-but-set-variable]
   96 |   auto tic_full_preint = utils::Timer::tic();
      |        ^~~~~~~~~~~~~~~
[... 130 lines cut ...]
   70 | static void getValidKeypointsFromStatusKeypointsCV(
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/include/kimera-vio/common/VioNavState.h:27,
                 from /mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/include/kimera-vio/backend/VioBackend-definitions.h:28,
                 from /mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/include/kimera-vio/frontend/FrontendInputPacketBase.h:17,
                 from /mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/include/kimera-vio/frontend/rgbd/RgbdImuSyncPacket.h:11,
                 from /mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/src/frontend/rgbd/RgbdImuSyncPacket.cpp:9:
/mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/include/kimera-vio/common/vio_types.h:70:13: warning: ‘void VIO::getValidKeypointsFromStatusKeypointsCV(const StatusKeypointsCV&, VIO::KeypointsCV*)’ defined but not used [-Wunused-function]
   70 | static void getValidKeypointsFromStatusKeypointsCV(
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/include/kimera-vio/backend/RegularVioBackend-definitions.h:28,
                 from /mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/include/kimera-vio/backend/RegularVioBackend.h:25,
                 from /mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/include/kimera-vio/backend/VioBackendModule.h:18,
                 from /mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/src/backend/VioBackendModule.cpp:16:
/mnt/C462F17762F16F14/work/SLAM/Kimera-VIO/include/kimera-vio/common/vio_types.h:70:13: warning: ‘void VIO::getValidKeypointsFromStatusKeypointsCV(const StatusKeypointsCV&, VIO::KeypointsCV*)’ defined but not used [-Wunused-function]
   70 | static void getValidKeypointsFromStatusKeypointsCV(
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[1]: *** [CMakeFiles/Makefile2:838: CMakeFiles/kimera_vio.dir/all] Error 2
make: *** [Makefile:146: all] Error 2

```

* GTSAM version used:4.1
* OpenCV version used: 3.3.1
* Operating system and version: Ubuntu 20.04

marcusabate, 2022-09-28:
Haven't seen this before, if it is still an issue I'd recommend uninstalling your source installation and making sure you purge any related files, and then installing using the package manager. A sanity check would be to build the docker image, but I doubt that would error out the same way.

Let me know if this is still happening.

closed 2022-11-29 by marcusabate

Janebek, 2023-09-27:
I met the same problem, you need to reinstall glog.

Check here for more details!  https://github.com/MIT-SPARK/Kimera-VIO/issues/123