MIT-SPARK/Kimera-VIO #193: update for ubuntu 20

JessicaBertolasi, 2022-09-14:
Hi, I'm trying to create a vio tracking using kimera-vio, but actually I'm working with ubuntu 20 on my pc, is it compatible in some way with this project? have you suggestions, in case it isn't, for vio algorithm compatible and similar to this kimera version?
Thank you and best regards

marcusabate, 2022-09-28:
It will work with a couple minor adjustments.

First, if you are building in ROS make sure to apply the focal patch in [mesh_rviz_plugins](https://github.com/MIT-SPARK/mesh_rviz_plugins/blob/master/ubuntu_focal.patch) using `git apply ubuntu_focal.patch`. 

If you are building native (without ROS), use opencv 3.4.2 instead of 3.3.1.

I am regularly testing Kimera in 20.04 using this dockerfile, it may be useful as a starting point:
```
# Use an official Python runtime as a parent image
FROM ubuntu:20.04

LABEL maintainer="mabate@mit.edu"

# To avoid tzdata asking for geographic location...
ENV DEBIAN_FRONTEND=noninteractive

# Set the working directory to /root
ENV DIRPATH /root/
WORKDIR $DIRPATH

#Install build dependencies
RUN apt-get update && apt-get install -y --no-install-recommends apt-utils
RUN apt-get update && apt-get install -y git cmake build-essential pkg-config

# Install xvfb to provide a display to container for GUI realted testing.
RUN apt-get update && apt-get install -y xvfb

# Install OpenCV for Ubuntu 20.04
RUN apt-get update && apt-get install -y \
      unzip \
      libjpeg-dev \
      libpng-dev \
      libtiff-dev \
      libvtk7-dev \
      libgtk-3-dev \
      libatlas-base-dev \
      gfortran

RUN git clone https://github.com/opencv/opencv.git
RUN cd opencv && \
      git checkout tags/3.4.2 && \
      mkdir build

RUN git clone https://github.com/opencv/opencv_contrib.git
RUN cd opencv_contrib && \
      git checkout tags/3.4.2

RUN cd opencv/build && \
      cmake -DCMAKE_BUILD_TYPE=Release \
      -DCMAKE_INSTALL_PREFIX=/usr/local \
      -D BUILD_opencv_python=OFF \
      -D BUILD_opencv_python2=OFF \
      -D BUILD_opencv_python3=OFF \
      -DOPENCV_EXTRA_MODULES_PATH=$DIRPATH/opencv_contrib/modules .. && \
      make -j$(nproc) install

# Install GTSAM
RUN apt-get update && apt-get install -y libboost-all-dev libtbb-dev
ADD https://api.github.com/repos/borglab/gtsam/git/refs/heads/develop version.json
RUN git clone https://github.com/borglab/gtsam.git
RUN cd gtsam && \
    git fetch && \
    git checkout e5866799dff48239573cdd84964180867e50edd2 && \
    mkdir build && \
    cd build && \
    cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF -DGTSAM_BUILD_TESTS=OFF -DGTSAM_BUILD_EXAMPLES_ALWAYS=OFF -DCMAKE_BUILD_TYPE=Release -DGTSAM_BUILD_UNSTABLE=ON -DGTSAM_POSE3_EXPMAP=ON -DGTSAM_ROT3_EXPMAP=ON -DGTSAM_TANGENT_PREINTEGRATION=OFF .. && \
    make -j$(nproc) install

# Install Open_GV
RUN git clone https://github.com/marcusabate/opengv
RUN cd opengv && git apply march_native_disable.patch && \
      mkdir build
RUN cd opengv/build && \
      cmake -DCMAKE_BUILD_TYPE=Release \
      -DCMAKE_INSTALL_PREFIX=/usr/local \
      -DEIGEN_INCLUDE_DIRS=$DIRPATH/gtsam/gtsam/3rdparty/Eigen \
      -DEIGEN_INCLUDE_DIR=$DIRPATH/gtsam/gtsam/3rdparty/Eigen .. && \
      make -j$(nproc) install

# Install DBoW2
RUN git clone https://github.com/dorian3d/DBoW2.git
RUN cd DBoW2 && \
      mkdir build && \
      cd build && \
      cmake .. && \
      make -j$(nproc) install

# Install RobustPGO
ADD https://api.github.com/repos/MIT-SPARK/Kimera-RPGO/git/refs/heads/master version.json
RUN git clone https://github.com/MIT-SPARK/Kimera-RPGO.git
RUN cd Kimera-RPGO && \
      mkdir build && \
      cd build && \
      cmake .. && \
      make -j$(nproc)

# Install glog, gflags
RUN apt-get update && apt-get install -y libgflags2.2 libgflags-dev libgoogle-glog0v5 libgoogle-glog-dev

# Install Pangolin
RUN apt-get update && apt-get install -y libgl1-mesa-dev libglew-dev
RUN git clone --depth 1 --branch v0.6 https://github.com/stevenlovegrove/Pangolin.git
RUN cd Pangolin && \
      mkdir build && \
      cd build && \
      cmake .. && \
      make -j$(nproc)

# Clean
RUN apt-get clean && \
      rm -rf /var/lib/apt/lists/*

# NOTE: dev only, don't push
COPY ./ Kimera-VIO
RUN cd Kimera-VIO && \ 
      mkdir build && \
      cd build && \
      cmake -DCMAKE_BUILD_TYPE=Release .. && \
      make -j$(nproc)
```

kanster, 2022-10-21:
Here are some tips that might be helpful if you want to compile Kimera locally:

- Please follow the gtsam version suggested in [Kimera-RPGO](https://github.com/MIT-SPARK/Kimera-RPGO) which is `686e16aaae26c9a4f23d4af7f2d4a504125ec9c3`. The commit number shown in [Kimera-VIO](https://github.com/MIT-SPARK/Kimera-VIO/blob/master/docs/kimera_vio_install.md) does NOT work for me;
- `ISAMParams` is changed and the `cacheLinearizedFactors` and `evaluateNonlinearError` are set as default, you may need to comment some of the lines `VioBackendParams.h`
- If you are using OpenCV 4.X version, you may need to include another `viz/viz3d.hpp` in visualization related header files

Good luck

closed 2022-11-29 by marcusabate

umbnich, 2023-03-15:
Hi, what is the proper way to proceed?