MIT-SPARK/Kimera-VIO #195: Having a error while installing OpenGV

keiro23, 2022-10-12:
**Description:**
I am suffering from one error right now. In the process of installing Opengv, the following error appears. Obviously I checked that the file exists in that path, can you tell me why it doesn't work? I've tried changing the permissions too, and it seems I've done everything I can.
Pls help....

**Command:**
```cmake .. -DEIGEN_INCLUDE_DIR=/home/tonirv/Code/gtsam/gtsam/3rdparty/Eigen -DEIGEN_INCLUDE_DIRS=/home/tonirv/Code/gtsam/gtsam/3rdparty/Eigen
# replace this line with the command(s) you used
```

**Console output:**
```CMake Error at modules/FindEigen.cmake:33 (file):
  file failed to open for reading (No such file or directory):

    /home/tonirv/Code/gtsam/gtsam/3rdparty/Eigen/Eigen/src/Core/util/Macros.h
Call Stack (most recent call first):
  modules/FindEigen.cmake:59 (_eigen3_check_version)
  CMakeLists.txt:57 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/park/tonirv/Code/opengv/build/CMakeFiles/CMakeOutput.log".

# remove this line and paste your console output HERE - no screenshots please

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* Kimera-Vio branch, tag or commit used
* GTSAM version used:
* OpenGV version used:
* OpenCV version used: type `opencv_version`
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* Did you change the source code? (yes / no):

keiro23, 2022-10-12:
The command was 
cmake .. -DEIGEN_INCLUDE_DIR=/home/tonirv/Code/gtsam/gtsam/3rdparty/Eigen -DEIGEN_INCLUDE_DIRS=/home/tonirv/Code/gtsam/gtsam/3rdparty/Eigen

keiro23, 2022-10-12:
Ubuntu version is 18.04, and I ran the commands written in installation.md as it was before the error occurred.

keiro23, 2022-10-13:
Gtsam version : 4.1.1
Opengv version : 1.0
opencv version : 3.4.2

kanster, 2022-10-21:
> The command was cmake .. -DEIGEN_INCLUDE_DIR=/home/tonirv/Code/gtsam/gtsam/3rdparty/Eigen -DEIGEN_INCLUDE_DIRS=/home/tonirv/Code/gtsam/gtsam/3rdparty/Eigen

Please use your local directory instead of Toni's.

Modify the directory `/home/tonirv/.../gtsam` correctly may help you solve the problem

closed 2022-11-29 by marcusabate