MIT-SPARK/Kimera-VIO #202: double free or corruption (out) Aborted(core dumped) when running stereoVIOEuroc.bash

ujasmandavia, 2023-01-13:
**Description:**
double free or corruption (out)
./scripts/stereoVIOEuroc.bash: line 109: 1707011 Aborted                 (core dumped) 

I'm getting this error while running stereoVIOEuroc.bash. I have changed the GTSAM syntax at certain place since I'm using the 4.2 version of GTSAM but I'm able to compile the code perfectly

**Command:**
```
bash ./scripts/stereoVIOEuroc.bash -p /home/ujas/Euroc/V1_01_easy

```

**Console output:**
```

double free or corruption (out)
./scripts/stereoVIOEuroc.bash: line 109: 1707011 Aborted                 (core dumped) $BUILD_PATH/stereoVIOEuroc --dataset_type="$DATASET_TYPE" --dataset_path="$DATASET_PATH" --initial_k=50 --final_k=10000 --params_folder_path="$PARAMS_PATH" --use_lcd="$USE_LCD" --vocabulary_path="$VOCABULARY_PATH/ORBvoc.yml" --flagfile="$PARAMS_PATH/flags/stereoVIOEuroc.flags" --flagfile="$PARAMS_PATH/flags/Mesher.flags" --flagfile="$PARAMS_PATH/flags/VioBackend.flags" --flagfile="$PARAMS_PATH/flags/RegularVioBackend.flags" --flagfile="$PARAMS_PATH/flags/Visualizer3D.flags" --logtostderr=1 --colorlogtostderr=1 --log_prefix=1 --v=0 --vmodule=Pipeline*=00 --log_output="$LOG_OUTPUT" --log_euroc_gt_data="$LOG_OUTPUT" --save_frontend_images=1 --visualize_frontend_images=1 --output_path="$OUTPUT_PATH"

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* Kimera-Vio branch, tag or commit used: 641576fd86bdecbd663b4db3cb068f49502f3a2c
* GTSAM version used: 4.2, the latest version
* OpenGV version used:
* OpenCV version used: type `opencv_version`
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* Did you change the source code? (yes / no): yes, changes it to my GTSAM version

bundle-adjuster, 2023-02-24:
Me as well. I had followed exactly what was given. The issue also crops up when building the tests. Not sure exactly which test build fails but the same error crops up.

bundle-adjuster, 2023-03-02:
The issue got fixed with the following commits and adding the following dependencies.

Gtsam - 686e16aaa
Note: Install libmetis-dev along with libboost-all-dev

Opencv - tags/3.3.1
Note: opencv_contrib takes same commit

RPGO - 8c5e163

I guess the issue crops up if you don't have libmetis installed.

marcusabate, 2023-08-21:
Closing because appears resolved, feel free to reopen if this is not the case.

closed 2023-08-21 by marcusabate