MIT-SPARK/Kimera-VIO #215: docker build fail : /usr/local/lib/libgtsam.so.4.3a0: undefined reference to `std::experimental::filesystem::v1::__cxx11::path::_M_split_cmpts()'

chowkamlee81, 2023-09-20:
**Description:**
Hi. Tried to build on a fresh Ubuntu 18.04 machine, with a fresh docker install.

**Command:**
```
sudo docker build --rm -t kimera_vio -f ./scripts/docker/Dockerfile . 
```

**Console output:**
```

 4.534 Scanning dependencies of target KimeraRPGO
#0 4.582 [ 50%] Building CXX object CMakeFiles/KimeraRPGO.dir/src/GenericSolver.cpp.o
#0 4.582 [ 50%] Building CXX object CMakeFiles/KimeraRPGO.dir/src/RobustSolver.cpp.o
#0 4.582 [ 56%] Building CXX object CMakeFiles/KimeraRPGO.dir/src/Logger.cpp.o
#0 4.582 [ 62%] Building CXX object CMakeFiles/KimeraRPGO.dir/src/utils/GraphUtils.cpp.o
#0 6.632 [ 68%] Linking CXX executable GenerateTrajectories
#0 6.746 /usr/local/lib/libgtsam.so.4.3a0: undefined reference to `std::experimental::filesystem::v1::__cxx11::path::_M_split_cmpts()'
#0 6.746 /usr/local/lib/libgtsam.so.4.3a0: undefined reference to `std::experimental::filesystem::v1::status(std::experimental::filesystem::v1::__cxx11::path const&)'
#0 6.746 /usr/local/lib/libgtsam.so.4.3a0: undefined reference to `std::experimental::filesystem::v1::__cxx11::path::_M_find_extension() const'
#0 6.746 /usr/local/lib/libgtsam.so.4.3a0: undefined reference to `std::experimental::filesystem::v1::__cxx11::path::parent_path() const'
#0 6.754 collect2: error: ld returned 1 exit status
#0 6.758 make[2]: *** [GenerateTrajectories] Error 1
#0 6.758 CMakeFiles/GenerateTrajectories.dir/build.make:105: recipe for target 'GenerateTrajectories' failed
#0 6.758 make[1]: *** [CMakeFiles/GenerateTrajectories.dir/all] Error 2
#0 6.758 make[1]: *** Waiting for unfinished jobs....
#0 6.758 CMakeFiles/Makefile2:178: recipe for target 'CMakeFiles/GenerateTrajectories.dir/all' failed
#0 22.24 [ 75%] Linking CXX shared library libKimeraRPGO.so
#0 22.40 [ 75%] Built target KimeraRPGO
#0 22.40 Makefile:140: recipe for target 'all' failed
#0 22.40 make: *** [all] Error 2
------
Dockerfile:82
--------------------
  81 |     RUN git clone https://github.com/MIT-SPARK/Kimera-RPGO.git
  82 | >>> RUN cd Kimera-RPGO && \
  83 | >>>       mkdir build && \
  84 | >>>       cd build && \
  85 | >>>       cmake .. && \
  86 | >>>       make -j$(nproc)
  87 |     

```

**Additional files:**
Please attach all the files needed to reproduce the error.

Please give also the following information:
* Kimera-Vio branch, tag or commit used
* GTSAM version used:
* OpenGV version used:
* OpenCV version used: type `opencv_version`
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* Did you change the source code? (yes / no):

sguttikon, 2023-10-16:
@chowkamlee81  I fixed the issue by using `Dockerfile_18_04`. 
`Kimera-VIO$ sudo docker build --rm -t kimera_vio -f ./Dockerfile_18_04 .`

However, i had to make changes to fix issues with [gtsam-build](https://github.com/borglab/gtsam/issues/1495), [ppa:deadsnakes/ppa](https://stackoverflow.com/questions/66597894/why-cannot-add-ppa-deadsnakes), [ffi.h](https://askubuntu.com/questions/1269492/ffi-h-no-such-file-or-directory-while-trying-to-install-docker-compose-on-ras) and [argon2-cffi-bindings](https://github.com/hynek/argon2-cffi-bindings/issues/8). The docker file modification looks as follows.
```
Kimera-VIO$ git diff Dockerfile_18_04
diff --git a/Dockerfile_18_04 b/Dockerfile_18_04
index c72cd035..d8434c78 100644
--- a/Dockerfile_18_04
+++ b/Dockerfile_18_04
@@ -47,6 +47,7 @@ RUN cd opencv/build && \
 RUN apt-get update && apt-get install -y libboost-all-dev libtbb-dev
 ADD https://api.github.com/repos/borglab/gtsam/git/refs/heads/master version.json
 RUN git clone https://github.com/borglab/gtsam.git
+RUN sed -i '2s/^/set(CMAKE_CXX_STANDARD_LIBRARIES "-lstdc++fs")\n/' ./gtsam/CMakeLists.txt
 RUN cd gtsam && \
     git fetch && \
     git checkout develop && \
@@ -87,7 +88,7 @@ RUN apt-get update && \
     apt-get install software-properties-common -y
 # Get python3
 RUN apt-get update && \
-    add-apt-repository ppa:deadsnakes/ppa
+    add-apt-repository 'ppa:deadsnakes/ppa'
 RUN apt-get update && \
     apt-get install -y python3.6 python3.6-dev python-pip python3-pip python-tk python3-tk
 RUN python3.6 -m pip install PyQt5==5.14
@@ -99,7 +100,8 @@ RUN git clone https://github.com/ToniRV/evo-1.git
 RUN cd evo-1 && python3.6 $(which pip3) install .
 
 # Install Kimera VIO Evaluation
-RUN python3.6 $(which pip3) install ipython prompt_toolkit
+RUN apt-get update && apt-get install -y libffi-dev
+RUN python3.6 $(which pip3) install ipython prompt_toolkit "setuptools_scm<7"
 # Hack to avoid Docker's cache when Kimera VIO Evaluation master branch is updated.
 ADD https://api.github.com/repos/MIT-SPARK/Kimera-VIO-Evaluation/git/refs/heads/master version.json
 RUN git clone https://github.com/MIT-SPARK/Kimera-VIO-Evaluation.git
```

HtutLynn, 2023-10-17:
@sguttikon , Hi, I am also having the same problem. If possible, can you provide the complete Dockerfile? I am still having trouble compiling Kimera-VIO.

sguttikon, 2023-10-17:
@HtutLynn This is my `./scripts/docker/Dockerfile`
```
# Use an official Python runtime as a parent image
FROM ubuntu:18.04

MAINTAINER Antoni Rosinol "arosinol@mit.edu"

# To avoid tzdata asking for geographic location...
ENV DEBIAN_frontend noninteractive

ENV TZ=America/New_York
RUN echo "Preparing geographic area ..."
RUN ln -snf /usr/share/zoneinfo/$TZ /etc/localtime && echo $TZ > /etc/timezone

# Set the working directory to /root
ENV DIRPATH /root
WORKDIR $DIRPATH

#Install build dependencies
RUN apt-get update && apt-get install -y --no-install-recommends apt-utils
RUN apt-get update && apt-get install -y git cmake

# Install xvfb to provide a display to container for GUI related testing.
RUN apt-get update && apt-get install -y xvfb

# Install GTSAM
RUN apt-get update && apt-get install -y libboost-all-dev
ADD https://api.github.com/repos/borglab/gtsam/git/refs/heads/master version.json
RUN git clone https://github.com/borglab/gtsam.git
RUN cd gtsam && \
    git fetch && \
    git checkout 4.1.1 && \
    mkdir build && \
    cd build && \
    cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DGTSAM_BUILD_TESTS=OFF -DGTSAM_BUILD_EXAMPLES_ALWAYS=OFF -DCMAKE_BUILD_TYPE=Release -DGTSAM_BUILD_UNSTABLE=ON -DGTSAM_TANGENT_PREINTEGRATION=OFF .. && \
    make -j$(nproc) install

# Install OpenCV for Ubuntu 18.04
RUN apt-get update && apt-get install -y \
      build-essential cmake unzip pkg-config \
      libjpeg-dev libpng-dev libtiff-dev \
      libvtk6-dev \
      libgtk-3-dev \
      libatlas-base-dev gfortran

RUN git clone https://github.com/opencv/opencv.git
RUN cd opencv && \
      git checkout tags/3.3.1 && \
      mkdir build

RUN git clone https://github.com/opencv/opencv_contrib.git
RUN cd opencv_contrib && \
      git checkout tags/3.3.1

RUN cd opencv/build && \
      cmake -DCMAKE_BUILD_TYPE=Release \
      -DCMAKE_INSTALL_PREFIX=/usr/local \
      -D BUILD_opencv_python=OFF \
      -D BUILD_opencv_python2=OFF \
      -D BUILD_opencv_python3=OFF \
      -DOPENCV_EXTRA_MODULES_PATH=$DIRPATH/opencv_contrib/modules .. && \
      make -j$(nproc) install

# Install Open_GV
RUN git clone https://github.com/laurentkneip/opengv
RUN cd opengv && \
      mkdir build
RUN cd opengv/build && \
      cmake -DCMAKE_BUILD_TYPE=Release \
      -DCMAKE_INSTALL_PREFIX=/usr/local \
      -DEIGEN_INCLUDE_DIRS=$DIRPATH/gtsam/gtsam/3rdparty/Eigen \
      -DEIGEN_INCLUDE_DIR=$DIRPATH/gtsam/gtsam/3rdparty/Eigen .. && \
      make -j$(nproc) install

# Install DBoW2
RUN git clone https://github.com/dorian3d/DBoW2.git
RUN cd DBoW2 && \
      mkdir build && \
      cd build && \
      cmake .. && \
      make -j$(nproc) install

# Install RobustPGO
ADD https://api.github.com/repos/MIT-SPARK/Kimera-RPGO/git/refs/heads/master version.json
RUN git clone https://github.com/MIT-SPARK/Kimera-RPGO.git
RUN cd Kimera-RPGO && \
      mkdir build && \
      cd build && \
      cmake .. && \
      make -j$(nproc)

## [Optional] Install Kimera-VIO-Evaluation from PyPI
RUN apt-get update && \
    apt-get install software-properties-common -y
# Get python3
RUN apt-get update && \
    add-apt-repository 'ppa:deadsnakes/ppa'
RUN apt-get update && \
    apt-get install -y python3.6 python3.6-dev python-pip python3-pip python-tk python3-tk
RUN python3.6 -m pip install PyQt5==5.14

# Install evo-1 for evaluation
# Hack to avoid Docker's cache when evo-1 master branch is updated.
ADD https://api.github.com/repos/ToniRV/evo-1/git/refs/heads/master version.json
RUN git clone https://github.com/ToniRV/evo-1.git
RUN cd evo-1 && python3.6 $(which pip3) install .

# Install spark_vio_evaluation
RUN apt-get update && apt-get install -y libffi-dev
RUN python3.6 $(which pip3) install ipython prompt_toolkit "setuptools_scm<7"
# Hack to avoid Docker's cache when spark_vio_evaluation master branch is updated.
ADD https://api.github.com/repos/ToniRV/Kimera-VIO-Evaluation/git/refs/heads/master version.json
RUN git clone https://github.com/ToniRV/Kimera-VIO-Evaluation.git
# We use `pip3 install -e .` so that Jinja2 has access to the webiste template...
RUN cd Kimera-VIO-Evaluation && python3.6 $(which pip3) install -e .

# Add credentials on build
ARG SSH_PRIVATE_KEY
RUN mkdir /root/.ssh/
RUN echo "${SSH_PRIVATE_KEY}" > /root/.ssh/id_rsa
RUN chmod 600 ~/.ssh/id_rsa

# Make sure your domain is accepted
RUN touch /root/.ssh/known_hosts
RUN ssh-keyscan -t rsa github.com >> /root/.ssh/known_hosts

# Install glog, gflags
RUN apt-get update && apt-get install -y libgflags2.2 libgflags-dev libgoogle-glog0v5 libgoogle-glog-dev

# Install Kimera-VIO
RUN git clone https://github.com/MIT-SPARK/Kimera-VIO.git
RUN cd Kimera-VIO && mkdir build && cd build && cmake .. && make -j$(nproc)

# Download and extract EuRoC dataset.
RUN apt-get update && apt-get install -y wget
RUN wget http://robotics.ethz.ch/~asl-datasets/ijrr_euroc_mav_dataset/vicon_room1/V1_01_easy/V1_01_easy.zip
RUN mkdir -p $DIRPATH/euroc && unzip V1_01_easy.zip -d $DIRPATH/euroc

# Yamelize euroc dataset
RUN bash $DIRPATH/Kimera-VIO/scripts/euroc/yamelize.bash -p $DIRPATH/euroc

#CMD xvfb-run $DIRPATH/Kimera-VIO/scripts/stereoVIOEuroc.bash -p $DIRPATH/euroc -r
CMD $DIRPATH/Kimera-VIO/scripts/stereoVIOEuroc.bash -p $DIRPATH/euroc -r
```

HtutLynn, 2023-10-17:
@sguttikon Thanks for the dockerfile. I am still getting erros compiling Kimera-VIO. I was wondering if you have ever encountered this issue before.

[image]

sguttikon, 2023-10-17:
@HtutLynn I didn't encountered this issue... a naive look feels like a gcc issue probably, sorry I can't say much in this regard.

marcusabate, 2023-11-03:
We are in the process of updating to the latest gtsam, which solves most of the docker issues. In the meantime, I'd recommend switching to the `feature/prerelease` branch and using one of those dockerfiles. 20.04 is the one we do active development with, that dockerfile is working well right now. The 18.04 dockerfile on that branch is also building on my machine right now.