MIT-SPARK/Kimera-VIO #218: How to run EUROC style custom dataset without ground truth estimate data

HtutLynn, 2023-10-26:
Hi, I have a dataset which is collected with iphone 13 and trying to use with Kimera-VIO. Since I have no idea on how to create a dataprovider by myself. I tried formatting the data from iphone13 as the same format as EuroC dataset.  However there is no ground truth estimate data for the custom dataset, created from iphone sensors so I tried modifying the `EurocDataProvider.cpp` as mentioned in https://github.com/MIT-SPARK/Kimera-VIO/issues/157 . However, a new error from GTSAM is causing the pipeline unable to run even on sample EUROC dataset. 

Here is the modified `EurocDataProvider.cpp`,

```
/* ----------------------------------------------------------------------------
 * Copyright 2017, Massachusetts Institute of Technology,
 * Cambridge, MA 02139
 * All Rights Reserved
 * Authors: Luca Carlone, et al. (see THANKS for the full author list)
 * See LICENSE for the license information
 * -------------------------------------------------------------------------- */

/**
 * @file   EurocDataProvider.h
 * @brief  Parse EUROC dataset.
 * @author Antoni Rosinol
 * @author Yun Chang
 * @author Luca Carlone
 */

#include "kimera-vio/dataprovider/EurocDataProvider.h"

#include <algorithm>  // for max
#include <fstream>
#include <map>
#include <string>
#include <utility>  // for pair<>
#include <vector>

#include <gflags/gflags.h>
#include <glog/logging.h>
#include <gflags/gflags.h>

#include <gtsam/base/Vector.h>
#include <gtsam/geometry/Cal3DS2.h>
#include <gtsam/geometry/Pose3.h>

#include "kimera-vio/frontend/StereoFrame.h"
#include "kimera-vio/imu-frontend/ImuFrontend-definitions.h"
#include "kimera-vio/logging/Logger.h"
#include "kimera-vio/utils/YamlParser.h"

DEFINE_string(dataset_path,
              "/Users/Luca/data/MH_01_easy",
              "Path of dataset (i.e. Euroc, /Users/Luca/data/MH_01_easy).");
DEFINE_int64(initial_k,
             0,
             "Initial frame to start processing dataset, "
             "previous frames will not be used.");
DEFINE_int64(final_k,
             100000,
             "Final frame to finish processing dataset, "
             "subsequent frames will not be used.");
DEFINE_bool(log_euroc_gt_data,
            false,
            "Log Euroc ground-truth data to file for later evaluation.");

namespace VIO {

/* -------------------------------------------------------------------------- */
EurocDataProvider::EurocDataProvider(const std::string& dataset_path,
                                     const int& initial_k,
                                     const int& final_k,
                                     const VioParams& vio_params)
    : DataProviderInterface(),
      dataset_path_(dataset_path),
      current_k_(std::numeric_limits<FrameId>::max()),
      initial_k_(initial_k),
      final_k_(final_k),
[... 795 lines cut ...]
  current_k_++;
  return true;
}

}  // namespace VIO

```

Here is the Error from GTSAM,

```
I1025 10:37:58.958392  4880 EurocDataProvider.cpp:804] Running dataset between frame 50 and frame 2912
terminate called after throwing an instance of 'gtsam::IndeterminantLinearSystemException'
  what():
Indeterminant linear system detected while working near variable
8646911284551352320 (Symbol: x0).

Thrown when a linear system is ill-posed.  The most common cause for this
error is having underconstrained variables.  Mathematically, the system is
underdetermined.  See the GTSAM Doxygen documentation at
http://borg.cc.gatech.edu/ on gtsam::IndeterminantLinearSystemException for
more information.
```

Am I doing something wrong with the data provider? Can you please provide guidance to make this work? Thanks in advance!

Mona-ShZeinoddin, 2023-11-03:
Hi 

Were you able to solve this?

marcusabate, 2023-11-03:
This error is probably not related to your data-provider. GTSAM throws these errors when the system is ill-posed, so in this case the optimization is likely under-constrained. This means that many of the factors in the optimization are not agreeing with each other. 

It could be caused by any of the factors. Some common problems: 
- Extrinsic calibration is wrong, so as the robot starts moving the vision and imu don't agree. 
- IMU gravity vector is not correct
- Vision factors (feature tracks) don't agree with each other (lots of occlusions, triangulation isn't correct), which can happen if you're using mono and don't have good visual features, or if you are using stereo and have a bad stereo calibration
- Prior factors are not accurate. This is relevant for the IMU especially, those biases are set in `BackendParams.yaml`.

Given your discussion in #217 it seems likely the two problems are related. I would check that IMU data is properly formed, that your gravity vector is correct, and that your extrinsic calibration between the IMU and the camera is correct.

As for why your code fails to run Euroc, I'm not sure. Have you set `autoInitialize` in `BackendParams.yaml` to 0? This will be required if you run Euroc without access to ground truth. One note: you've commented out every call to `CHECK_GT`, this is not a call to any ground truth function but rather sanity checks to make sure the incoming data is correct. You can keep most or all of those uncommented. 

More generally, I'd recommend making a new `iPhoneDataProvider` instead of modifying the EurocDataProvider directly. You can copy most of the code and not include any GT parts, and you can retain full Euroc functionality. You'll just need to add your new DataProvider to the options in `examples/KimeraVIO.cpp` and use the `dataset_type` variable to instruct the executable to use your new DataProvider by assigning it to value 2 or similar.

HtutLynn, 2023-11-07:
@marcusabate , Hi, as for the GTSAM error. I used the modified `EurocDataProvider.cpp` to test Mono VIO pipeline with EUROC dataset and I am not using custom dataset from iPhone sensors. To test whether Kimera works properly without ground truth data with modified script, I just deleted `ground_truth_estimates`  from `V1_01_easy` folder but this GTSAM error comes up. So I think it doesn't have to do with the dataset but the script that I modified. `autoinitialize` is also set to `0`.

HtutLynn, 2023-12-28:
@cdebeunne , Thanks, it fixed my problem.

closed 2023-12-28 by HtutLynn