MIT-SPARK/Kimera-VIO #219: Dockerfile bug

zhuoyuan00, 2023-10-27:
**Description:**
As the depencies upgrades, the dockerfile can be no longer used.
Mistakes occurs begins from installation of Kimera-RPGO because of devel branch of gtsam. And a lot of mistakes follows beside this.
**Command:**
```
Could you please offer a new dockerfile or share your dockerhub environment?
```

**Console output:**
```
 => ERROR [23/38] RUN cd Kimera-RPGO &&       mkdir build &&       cd build &&       cmake .. &&       make -j$(nproc)                                   16.5s

/usr/local/lib/libgtsam.so.4.3a0: undefined reference to `std::experimental::filesystem::v1::__cxx11::path::_M_split_cmpts()'
4.214 /usr/local/lib/libgtsam.so.4.3a0: undefined reference to `std::experimental::filesystem::v1::status(std::experimental::filesystem::v1::__cxx11::path const&)'
4.214 /usr/local/lib/libgtsam.so.4.3a0: undefined reference to `std::experimental::filesystem::v1::__cxx11::path::_M_find_extension() const'
4.214 /usr/local/lib/libgtsam.so.4.3a0: undefined reference to `std::experimental::filesystem::v1::__cxx11::path::parent_path() const'

```

Please give also the following information:
As the dockerfile includes.

* Kimera-Vio branch, tag or commit used
* GTSAM version used:
* OpenGV version used:
* OpenCV version used: type `opencv_version`
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10): 
* Did you change the source code? (yes / no):

marcusabate, 2023-11-03:
We are in the process of updating the gtsam version, in the meantime I'd recommend using the 20.04 dockerfile. It's on the `feature/prerelease` branch, which will be merged to master soon. The 18.04 dockerfile on that branch is working on my machine right now as well.

zhuoyuan00, 2023-11-30:
> We are in the process of updating the gtsam version, in the meantime I'd recommend using the 20.04 dockerfile. It's on the `feature/prerelease` branch, which will be merged to master soon. The 18.04 dockerfile on that branch is working on my machine right now as well.

I have tried 
'git checkout  feature/prerelease'
and then 
'docker build --rm -t kimera_vio -f ./scripts/docker/Dockerfile . '

However, mistakes also occurs while it is installing RPGO.
```
4.100 [ 68%] Linking CXX executable GenerateTrajectories
4.161 /usr/local/lib/libgtsam.so.4.3a0: undefined reference to `std::experimental::filesystem::v1::__cxx11::path::_M_split_cmpts()'
4.161 /usr/local/lib/libgtsam.so.4.3a0: undefined reference to `std::experimental::filesystem::v1::status(std::experimental::filesystem::v1::__cxx11::path const&)'
4.161 /usr/local/lib/libgtsam.so.4.3a0: undefined reference to `std::experimental::filesystem::v1::__cxx11::path::_M_find_extension() const'
4.161 /usr/local/lib/libgtsam.so.4.3a0: undefined reference to `std::experimental::filesystem::v1::__cxx11::path::parent_path() const'
4.161 collect2: error: ld returned 1 exit status
4.162 make[2]: *** [GenerateTrajectories] Error 1
4.162 CMakeFiles/GenerateTrajectories.dir/build.make:105: recipe for target 'GenerateTrajectories' failed
4.162 CMakeFiles/Makefile2:178: recipe for target 'CMakeFiles/GenerateTrajectories.dir/all' failed
4.162 make[1]: *** [CMakeFiles/GenerateTrajectories.dir/all] Error 2
4.162 make[1]: *** Waiting for unfinished jobs....
16.18 [ 75%] Linking CXX shared library libKimeraRPGO.so
16.25 [ 75%] Built target KimeraRPGO
16.25 Makefile:140: recipe for target 'all' failed
16.25 make: *** [all] Error 2
------
Dockerfile:79
--------------------
  78 |     RUN git clone https://github.com/MIT-SPARK/Kimera-RPGO.git
  79 | >>> RUN cd Kimera-RPGO && \
  80 | >>>       mkdir build && \
  81 | >>>       cd build && \
  82 | >>>       cmake .. && \
  83 | >>>       make -j$(nproc)
  84 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c cd Kimera-RPGO &&       mkdir build &&       cd build &&       cmake .. &&       make -j$(nproc)" did not complete successfully: exit code: 2
```
Could you help me with it? Thanks very very much.

zhuoyuan00, 2023-12-01:
This question has been solved by
'docker build --rm -t kimera_vio -f ./Docerfile_18_04 . ' and kimera_vio has been installed successfully in this docker environment.
However, while running Kimera-VIO in Euroc's dataset, error occurs and it shows
```
W1201 07:34:17.685111 33490 Mesher.cpp:1377] Missing landmark information to build 3D Mesh.
E1201 07:34:17.685676 33490 Mesher.cpp:1729] Delaunay Triangle out of image (size: x: 0, y: 0, height: 480, width 752
 Triangle: x, y: 
2256, 0
0, 2256
639.865, 431.495
W1201 07:34:17.688410 33491 OpenCvVisualizer3D.cpp:1130] No landmark information for Visualizer. Not displaying 3D points.
W1201 07:34:17.691613 33487 OpenCvDisplay.cpp:143] Missing Mesh in visualization's 3D widgets.
W1201 07:34:17.711465 33488 StereoMatcher.cpp:130] sparseStereoMatching: StereoFrame is already rectified!
W1201 07:34:17.711479 33488 StereoCamera.cpp:263] Rectifying already rectified stereo frame ...
I1201 07:34:17.722695 33489 RegularVioBackend.cpp:212] Tracker has a LOW_DISPARITY status.
W1201 07:34:17.723078 33490 Mesher.cpp:1377] Missing landmark information to build 3D Mesh.
W1201 07:34:17.727409 33491 OpenCvVisualizer3D.cpp:1130] No landmark information for Visualizer. Not displaying 3D points.
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
X Error of failed request:  GLXBadContext
  Major opcode of failed request:  152 (GLX)
  Minor opcode of failed request:  6 (X_GLXIsDirect)
  Serial number of failed request:  65
  Current serial number in output stream:  64
/root/Documents/Kimera-VIO/scripts/stereoVIOEuroc.bash: line 110: 33487 Segmentation fault      (core dumped) $BUILD_PATH/stereoVIOEuroc --dataset_type="$DATASET_TYPE" --dataset_path="$DATASET_PATH" --initial_k=50 --final_k=10000 --params_folder_path="$PARAMS_PATH" --use_lcd="$USE_LCD" --vocabulary_path="$VOCABULARY_PATH/ORBvoc.yml" --flagfile="$PARAMS_PATH/flags/stereoVIOEuroc.flags" --flagfile="$PARAMS_PATH/flags/Mesher.flags" --flagfile="$PARAMS_PATH/flags/VioBackend.flags" --flagfile="$PARAMS_PATH/flags/RegularVioBackend.flags" --flagfile="$PARAMS_PATH/flags/Visualizer3D.flags" --logtostderr=1 --colorlogtostderr=1 --log_prefix=1 --v=0 --vmodule=Pipeline*=00 --log_output="$LOG_OUTPUT" --log_euroc_gt_data="$LOG_OUTPUT" --save_frontend_images=1 --visualize_frontend_images=1 --output_path="$OUTPUT_PATH"
```
What's the problem please?