MIT-SPARK/Kimera-VIO #223: Kimera VIO cmake build error.

threedee003, 2024-01-05:
Dear @marcusabate ,
  I followed the installation instructions and tried building the package. But I am getting the following error. Can you please tell me, what do I need to do to resolve this ? I would be happy to provide you any additional information.

```
[ 92%] Building CXX object CMakeFiles/testKimeraVIO.dir/tests/testInitializationFromImu.cpp.o
[ 93%] Building CXX object CMakeFiles/testKimeraVIO.dir/tests/testVioBackend.cpp.o
[ 93%] Building CXX object CMakeFiles/testKimeraVIO.dir/tests/testVioBackendParams.cpp.o
[ 94%] Building CXX object CMakeFiles/testKimeraVIO.dir/tests/testVioParams.cpp.o
[ 95%] Building CXX object CMakeFiles/testKimeraVIO.dir/tests/testVisionImuFrontendParams.cpp.o
[ 95%] Building CXX object CMakeFiles/testKimeraVIO.dir/tests/testFeatureDetectorParams.cpp.o
[ 96%] Building CXX object CMakeFiles/testKimeraVIO.dir/tests/testFeatureDetector.cpp.o
[ 96%] Building CXX object CMakeFiles/testKimeraVIO.dir/tests/testVisualizer3D.cpp.o
[ 97%] Building CXX object CMakeFiles/testKimeraVIO.dir/tests/testOnlineAlignment.cpp.o
[ 98%] Building CXX object CMakeFiles/testKimeraVIO.dir/tests/testOpticalFlowPredictor.cpp.o
[100%] Linking CXX executable stereoVIOEuroc
/usr/bin/ld: warning: libopencv_core.so.409, needed by /usr/local/lib/libDBoW2.so, not found (try using -rpath or -rpath-link)
[100%] Built target stereoVIOEuroc
[100%] Linking CXX executable testKimeraVIO
/usr/bin/ld: warning: libopencv_core.so.409, needed by /usr/local/lib/libDBoW2.so, not found (try using -rpath or -rpath-link)
/home/bikram/Kimera-VIO/build/testKimeraVIO: error while loading shared libraries: libopencv_core.so.409: cannot open shared object file: No such file or directory
CMake Error at /usr/share/cmake-3.16/Modules/GoogleTestAddTests.cmake:40 (message):
  Error running test executable.

 Path: '/home/bikram/Kimera-VIO/build/testKimeraVIO'
 Result: 127
 Output:

make[2]: *** [CMakeFiles/testKimeraVIO.dir/build.make:1011: testKimeraVIO] Error 1
make[2]: *** Deleting file 'testKimeraVIO'
make[1]: *** [CMakeFiles/Makefile2:911: CMakeFiles/testKimeraVIO.dir/all] Error 2
make: *** [Makefile:141: all] Error
```

regards,
Tribikram.

yangzhongii, 2024-04-29:
I encountered the same problem as you, how did you solve it?

marcusabate, 2024-04-29:
Sometimes this happens if you make changes to the code and try to rebuild, though that doesn't seem to be the case here. Try first doing `catkin clean kimera_vio kimera_vio_ros` and then `catkin build kimera_vio` to see if the clean and rebuild works (this is usually it for me).
It's also giving a warning about libopencv_core.so that I've not seen before, are you building `opencv3_catkin` in the same workspace? Try building it before Kimera, it shouldn't matter but that might help. Alternatively, install a system-wide version of opencv.

yangzhongii, 2024-05-12:
Hi, Abate
Sorry for being late to answer your question, I am recently busy on my university graduation. When I try to rebuild it, I did meet some problems, for example the Gstam. But I have never met any warning about libopencv_core.so. I am wondering if it is convenient for you to send me your error about libopencv. Thanks