MIT-SPARK/Kimera-VIO #227: Running VIO with own camera and IMU data + acquiring the pose

scode-cmd, 2024-02-12:
Hello,

This questions might seem silly, but i've been banging my head against the wall trying to figure it out. Basically, I want to create a VIO stereo pipline that takes camera images + IMU data acquired in realtime, and _not from a dataset_, then get the 3d poses from the resulting calculation. I'm not using ROS and not intending to use it.

1. Do I need custom data provider in that case?
2. After we spin the VIO pipline with our own (synchronized) data, how do we acquire the latest 3d pose from the VIO pipline? I'm having hard time understanding the interaction between frontend and backend.

tbrayer2, 2024-03-12:
I've been straggling too. For simplicity - I also prepared a sequence with Carla simulator, where I have a full control over everything. I put the camera and IMU at the exact same location with the same orientation, yet - I can't even execute it on this synthetic sequence.

zippyzoo77, 2024-07-03:
Any luck on this for either of you so far? I'm having troubles with the ROS portion producing garbage results, and am hoping to do something similar to what you're talking about. I think I will need a custom data provider, I'm just not sure how to start on getting it to loop properly each time an image/batch of imu measurements is put into the queue rather than expecting a static dataset.

tbrayer2, 2024-07-09:
Hi,

No. We stopped using it for now.