MIT-SPARK/Kimera-VIO #229: Can I compile Kimera without the OpenCV viz module?

sarika93, 2024-03-05:
When I compile Kimera-VIO, I get the below error, indicating that the OpenCV viz module is not installed.

`[  6%] Building CXX object CMakeFiles/kimera_vio.dir/src/dataprovider/EurocDataProvider.cpp.o
In file included from /cc_kimera/Kimera-VIO/include/kimera-vio/logging/Logger.h:25:0,
                 from /cc_kimera/Kimera-VIO/include/kimera-vio/dataprovider/EurocDataProvider.h:36,
                 from /cc_kimera/Kimera-VIO/src/dataprovider/EurocDataProvider.cpp:17:
/cc_kimera/Kimera-VIO/include/kimera-vio/mesh/Mesh.h:23:10: fatal error: opencv2/viz/types.hpp: No such file or directory
 #include <opencv2/viz/types.hpp>  // Just for color type.
          ^~~~~~~~~~~~~~~~~~~~~~~`

I do not want to install or use this module (as well as VTK) as I'm trying to run this on a not-especially-powerful aarch64 device and I don't really care for this type of visualization. 

Is there an easy way (e.g. a flag) to disable this as a build dependency? If not, how important is it for Kimera's core VIO operation and do would it be simple to remove (for my use case)?

bharadwajsirigadi, 2024-06-09:
Did you find the solution, i ran to same error.

sarika93, 2024-06-10:
There isn't an easy way to remove the VTK dependency, unfortunately. My workaround was to actually modify the code remove everything related to it from my build, which I have on a branch [here](https://github.com/Virtana/Kimera-VIO/tree/sarika/cross-compiling/docker). It's very hacky and based off of an older commit than master, but I really didn't want to install a bunch of unnecessary (for my application) things on my device. 

Are you compiling or cross-compiling? If you're compiling on a machine which has a package manager, the other solution would be to install VTK and recompile opencv from source. You can use the Dockerfile [here](https://github.com/Virtana/Kimera-VIO/blob/sarika/cross-compiling/docker/scripts/docker/Dockerfile#L45) as a guide for how to do this, but this is for Ubuntu 18.04, so you may need to make some changes for your device. I used opencv 4.6, rather than 3.3.1 which is included there.