MIT-SPARK/Kimera-VIO #239: Ubuntu 22.04 Build not Recognizing Viz Module

CharlieRusso001, 2024-08-28:
**Description:**
When compiling the viz module from OpenCV is not recognized.

**Command:**
```
# replace this line with the command(s) you used
```
make -j$(nproc)

**Console output:**
```

/home/charl/scald/mapping/Kimera-VIO/src/frontend/RgbdCamera.cpp: In function ‘void VIO::convertToPcl(const cv::Mat&, const cv::Mat&, const Intrinsics&, const T&, cv::Mat*, cv::Mat*)’:
/home/charl/scald/mapping/Kimera-VIO/src/frontend/RgbdCamera.cpp:53:48: error: ‘viz’ is not a member of ‘cv’
   53 |       cv::Mat(img_rows, img_cols, CV_8UC3, cv::viz::Color::red());

```

Please give also the following information:
* Kimera-Vio branch, tag or commit used: The most recent
* GTSAM version used: 
* OpenGV version used:
* OpenCV version used: type `opencv_version` 4.6.0
* Operating system and version (e.g. Ubuntu 16.04 or Windows 10):  Ubuntu 22.04
* Did you change the source code? (yes / no):  no

Is there any way to fix this build issue

limshoonkit, 2024-09-09:
Did you build your opencv from source?
1. If so you may need to enable the -DWITH_VTK=On option when building it.
2. You could also add the following line in the CMakeLists.txt.
```
find_package(OpenCV 4.10 REQUIRED 
            COMPONENTS 
            viz)
```
3. In the include folder for camera.h, add the following header file
```
#include <opencv2/viz.hpp>
```

Happy debugging!

uni-paul-taylor2, 2025-01-15:
I installed opencv using apt packages, following guidance from the [Dockerfile](https://github.com/MIT-SPARK/Kimera-VIO/blob/master/scripts/docker/Dockerfile) and I have the same issue on Ubuntu 22.04 as well :(

All the other dependencies install and Kimera-VIO itself has been the area of issue

uni-paul-taylor2, 2025-01-16:
Downgrading to Ubuntu 20.04 seems to work, using the exact same steps to attempt to build Kimera-VIO in 22.04, reproducible from [this commit](https://github.com/uni-paul-taylor2/kimera-vio-implementation/commit/7f0887e3a573a9d6d57d1d3e25b5b03af235b718)

d3276605, 2025-02-15:
Found a fix-

First, you will need to build opencv with the command "DWITH_VTK=On"
```
cmake -DWITH_VTK=On ..
make make -j4
```

Then add this to the top of the files 
`src/frontend/RgbdCamera.cpp`
`include/kimera-vio/mesh/MeshOptimization.h`

```
#include <opencv2/viz/viz3d.hpp>
#include <opencv2/viz/types.hpp>
```

And also, it looks like the current `tests/testTracker.cpp` file also has an error. Replace the file with the following
https://gist.github.com/d3276605/bf93a74e667c2fd9ec78b4c352c8831f

AnachronicNomad, 2025-02-16:
Can confirm this behavior occurs on multiple aarch64 platforms.  Tested with all dependencies built (correctly) from source.

AnachronicNomad, 2025-02-17:
Also needs to have `#include <vector>` included in `include/kimera-vio/utils/UtilsNumerical.h`

AnachronicNomad, 2025-02-18:
@d3276605 I was able to successfully build the library and install after applying the above fixes, and disabling the tests with a `cmake -D` option, (specifically, I turned off tests using the `ccmake` TUI.