MIT-SPARK/Kimera-VIO #254: EurocMono params on master gives poor VIO performance

sarika93, 2025-02-13:
**Description:**
Running the master branch using the stereoVIOeuroc.bash script with Euroc V1_01_easy dataset using the default EurocMono params on master gives very poor VIO performance. The pipeline runs until completion. **Note: stereo performance is still good** at less than +/-0.1m of error in each axis with the same dataset.

The below shows the error performance with the default EurocMono setup:

[image]

This was not the case in an older version of master (git hash: a2cf0af0f3938ae694254d159234001b7291d2f6), which gave the following error performance.

[image]

It looks like with OpenCV or GTSAM related updates something was broken. 

However, I was able to find a working config which gives good VIO performance when running with master. Changing the frontend detector from GFTT (`feature_detector_type: 3` in `FrontendParams.yaml`) to FAST (`feature_detector_type: 0` in `FrontendParams.yaml`) , the performance improves signficantly. See below:

[image]

Changing the linearization mode from Hessian to Implicit Schur (`linearizationMode: 1` in `BackendParams.yaml`) further improves performance:

[image]

All graphs were generated using [this](https://github.com/Virtana/indoor_drone_experimentation/blob/main/kimera_remote_eval/drift_assessment.py) script.

**I will be submitting a PR with these updated params for the EurocMono configuration**. However, this doesn't fix any possible underlying issues here.

**Steps to reproduce error:**
1. Checkout this repository to the latest version of master
2.  Update the `scripts/stereoVIOeuroc.bash` file to use EurocMono params and point to the relevant dataset path. Output should also be logged. 
3. Run `scripts/stereoVIOeuroc.bash`. Using the visualizer, the explosion in pose estimation should be easily visible. Alternatively, the script linked above can be used to produce an error graph. 

**My testing setup**
* Kimera-Vio branch, tag or commit used: master, git hash: acf45cb9fef6caecffd9b4b953daa6a82686e897
* GTSAM version used: 4.2
* OpenGV version used:
* OpenCV version used: 4.2.0
* Operating system and version :  Ubuntu 20.04 via Docker
* Did you change the source code? (yes / no): No, this was tested on the master branch.

Vishnu-AP, 2025-03-24:
Hey @sarika93 !
I am facing very poor VIO performanc issues with the Stereo RealSense modules as well! I tried changing the front end parameters (GFTT / ORB / FAST) but I did not see any significant performance improvements, my ground truth movement is about 6m while the VIO estimates only 4m of movement. I was looking forward to trying out some of the parameter changes you made so please let me know if there are any changes that you made apart from the ones in the PR.
Thanks!

sarika93, 2025-03-24:
@Vishnu-AP can you try changing the `autoInitialize` parameter in BackendParams.yaml from 0 to 1? This should enable using IMU for initialization.

Hopefully this helps you resolve you issues.

Vishnu-AP, 2025-04-01:
@sarika93 ```autoInitialize``` Parameter was already set to 1, The VIO performs well in a more structured environment and fails to work well in places like large halls, long hallways etc. Do you know of any instances where VIO estimates were good even in such places?

Vishnu-AP, 2025-04-03:
@sarika93 Yes, I did have a look actually and the features tracked are very few so I am suspecting that could be an issue.